particle-iot-archived/docs-old #293: spark.github.io is pointing at new/old docs?

dmiddlecamp, 2015-02-17:


joegoggins, 2015-02-17:
spark.github.io is linked to [from here](https://www.spark.io/resources) under the "open source" link.  I think that's the intended behavior.  Is there something in particular that is wrong with this link/page?

dmiddlecamp, 2015-02-17:
Oh sorry, I was comparing this to this:

http://docs.spark.io/

http://spark.github.io/docs/#/

joegoggins, 2015-02-17:
Yea, gotcha, that is weird. 

@zsup , it looks like you were the last fellow to push to https://github.com/spark/spark.github.io, which is presumably where outdated the http://spark.github.io/docs/# are coming from. Do you know how to drop that docs content?

zsup, 2015-02-18:
Actually it was coming from `gh-pages`, which was how the docs site used to be published (now it's published via S3). I just deleted the `gh-pages` branch, so it's gone now

closed 2015-02-18 by zsup