particle-iot-archived/docs-old #302: downsize all images to be mobile friendly

kennethlimcp, 2015-03-06:
There are many high-resolution images (which are fantastic) but might not be mobile friendly in terms of page loading.

Might want to have a lower resolution / size image for the "mobile version" or simply lower the resolution in general to suit web/mobile land

technobly, 2015-03-06:
I think what could be a good way to do it is a decent sized image that's compressed as JPG, that serves as the "thumbnail" but not necessarily tiny like a thumbnail.  And that links to their larger high res images, possibly PNGs.  And then we document like this:

`[[image]](high-res-url)`

kennethlimcp, 2015-03-07:
@technobly, that would definitely be rad.

The last time i heard, there's going to be a docs overhaul but haven't been told who is officially in charge. Wouldn't want to spend time doing double work here ;)

closed 2015-05-31 by kennethlimcp