particle-iot-archived/docs-old #304: document publish rate limits in firmware spark.publish documentation?

dmiddlecamp, 2015-03-08:
I didn't see this, is this documented somewhere?

kennethlimcp, 2015-03-08:
@dmiddlecamp, it is not mentioned somewhere but i thought there's some discussion on the limits being lifted?

dmiddlecamp, 2015-03-09:
I'm always pushing to raise the limits when it makes sense, but in the meantime we should at least document what's in place now so people aren't caught offguard.  We've been talking about better feedback mechanisms for when things are rate limited, so it's more clear what's happening.

kennethlimcp, 2015-03-09:
@dmiddlecamp,

just to clarify before i send in a PR.

1.) No more than 1 `Spark.publish()` per second 

2.) Bursts of up to 4 per second?

dmiddlecamp, 2015-03-09:
Heya @kennethlimcp,

  That's about right, a rate of about 1 a second, with bursts of up to 4 allowed from the firmware.

Thanks!
David

kennethlimcp, 2015-03-09:
Here's the PR!

closed 2015-03-18 by kennethlimcp

technobly, 2015-03-20:
Thought I'd mention if you burst 4 messages back to back, it takes 4 seconds to recover before you can send 1 per second again.