particle-iot-archived/docs-old #310: Wrong value in Time.zone() example

pcamp, 2015-04-04:
The comment says "Set time zone to Eastern USA daylight saving time", so it should be `Time.zone(-4);` rather than `Time.zone(-5);`.

Reference: http://www.timeanddate.com/time/zones/edt

kennethlimcp, 2015-04-04:
@pcamp i don't live in US so i'm not sure about this but the idea is that you can actually adjust the time zone offset (+/-) from UTC.

Does the Eastern daylight time saving time change or fixed to -4? I'm just guessing it depends on which parts of the year? :smile_cat:

pcamp, 2015-04-04:
Eastern Daylight Time is always -4, but it is only used for part of the year.  During the rest of the year, the Eastern USA uses Eastern Standard Time, which is always -5.

kennethlimcp, 2015-04-04:
Sounds good! I shall change this. @technobly, can you just confirm this daylight time change is correct?

towynlin, 2015-04-06:
Thanks for pointing out this error @pcamp —yes @kennethlimcp :+1: this comment correctly says that EDT is always UTC-4 (only in use during the summer months), while the same part of the U.S. uses EST (UTC-5) for the winter months. And thanks Kenneth for getting clarification; time zones can definitely be confusing. :smile:

kennethlimcp, 2015-04-07:
@pcamp and @towynlin, thanks! Here's the fix as promised.

closed 2015-04-07 by kennethlimcp