particle-iot-archived/docs-old #315: Functions in wrong category

kennethlimcp, 2015-04-09:
Seems like we are placing:

```
Spark.deviceID()
Spark.sleep()
```

under **cloud functions** which totally does not reflect its purpose

zsup, 2015-04-10:
actually what that highlights to me is that these should probably not be part of the `Spark` object, but rather the `System` object. @m-mcgowan, thoughts?

kennethlimcp, 2015-04-13:
Changed has been implemented pending next firmware update: https://github.com/spark/firmware/issues/390

kennethlimcp, 2015-04-14:
Here's what i am proposing:

- Rename `Advanced` to `System` (http://docs.spark.io/firmware/#advanced)
- Move `System.reset()` to new `System` category (http://docs.spark.io/firmware/#libraries-system)
- Move `Spark.deviceID()` and `Spark.sleep()` to new `System` category (This will change with the renaming from Spark to System)
- Move `Spark.syncTime()` to after `Spark.process()` (currently at the bottom) (http://docs.spark.io/firmware/#spark-synctime)

zsup, 2015-04-14:
ping @m-mcgowan

m-mcgowan, 2015-04-14:
Sorry, for some reason not getting these pings. (Or they're buried under a zillion other notifications from github.)

The `reset()`, `deviceID()` and `sleep()` functions have been moved in the private repo to `System`. They are still available from `Spark` with a deprecation warning.

kennethlimcp, 2015-04-14:
Here's the PR: https://github.com/spark/docs/pull/318

Will merge in tonight if there's not objections/votes.

kennethlimcp, 2015-04-17:
Fixed in: https://github.com/spark/docs/pull/318

closed 2015-04-17 by kennethlimcp