particle-iot-archived/docs-old #317: Core wakeup pin behavior

kennethlimcp, 2015-04-14:
We need to explain clearer as to the behavior of the wakeup pin of the core:

Full details here: https://community.spark.io/t/having-trouble-using-sleep-with-wakeuppin/10691/45

kennethlimcp, 2015-04-16:
Possible docs:

```
edgeTriggerMode: defines when the interrupt should be triggered. 
Three constants are predefined as valid values. 
RISING to trigger when the pin goes from LOW to HIGH,
FALLING for when the pin goes from HIGH to LOW,
CHANGE to trigger the interrupt whenever the pin changes value.

In order to detect the desired trigger edge `Spark.sleep()` implicitly attaches appropriate pull resistors.
RISING uses pull-down
FALLING uses pull-up

CHANGE places the pin in tri-state or "floating input" mode, thus requiring the user to define an external circuit that forces a high or low state (pull-up, pull-down, driven by logic gate, etc.) in order to prevent erroneous triggers due to noise.

Any previous `pinMode()` setting for the designated `wakeUpPin` will be overridden by the call to `Spark.sleep(uint16_t wakeUpPin, uint16_t edgeTriggerMode [, uint32_t seconds = 0])`.
```

```
The optional third parameter acts as with `Spark.sleep(int seconds)` or `Spark.sleep(SLEEP_MODE_DEEP, int seconds)`, to wake the Core after some seconds automatically.
```

closed 2015-09-24 by kennethlimcp

m-mcgowan, 2015-09-25:
I recently removed the wakeup pin in the Core since it was preventing A0 from being used as a regular digital GPIO.  We can re-instate when there is time to investigate!