particle-iot-archived/docs-old #325: Add more description to SYS MODEs

kennethlimcp, 2015-05-02:
**AUTOMATIC**
- user code does not run when cloud connection/wifi drops

etc etc...

ScruffR, 2015-05-03:
In SEMI_AUTOMATIC mode the action of establishing a cloud connection falls apart in several "independent" steps (unlike the atomic action in AUTOMATIC mode).
In between each step, user code can be placed.

What are these sub-actions, how long (aprox./avg.) do they take and what are the respective color codes for each step?

Related topics
http://community.spark.io/t/create-time-out-for-spark-connect/10351
http://community.spark.io/t/system-mode-wifi-functions/11569/14

kennethlimcp, 2015-05-03:
@ScruffR ,

the thread you posted looks more like a code fix than a docs fix (though a docs fix would explain what might happen to the user but the former is the way to go)

1.) The colour scheme follows the existing. Blinking green during wifi connection and Blinking cyan during cloud connection.  Don't think we need to emphasize this more.

2.) For sub-actions, 

In **SEMI_AUTOMATIC** you can simply call `Spark.connect()` and the core would start to connect to Wifi and then the cloud. There's not time-out for this so the user has to perform a `Spark.disconnect()` using a timer loop. So as mentioned, a firmware fix would be better so that users can skip doing this manually.

ScruffR, 2015-05-03:
There needs to be a docs fix, since the docs do state `Spark.connect()` does block user code untill connected, which is ***not*** true, and my sample just shows that it isn't.
Originally I've written this to take care of the blocking connect, but then I realized it's not at all blocking (the disconnect interrupt never fires, not even without any WiFi available, but user code keeps blinking the LED).

As for sub-actions:
I know that you only need to call `Spark.connect()`, but implicitly it roughly falls into steps that you would manually achieve by sequentially calling `WiFi.on()`, `WiFi.connect()`, check for `WiFi.connecting()` and `WiFi.ready() then actually `Spark.connect()` and wait till `Spark.connected()` becomes true.

And as I see it, in between each or at least some of these steps `SPARK_WLAN_Loop()` or `Spark.process()` does come back to the user code - similar to an FSM. 
So it's not blocking.

And exactly this is something that needs to be doced by someone who actually knows it off by heart.

As for the color scheme, where exactly could I find e.g. breathing green, rapidly flashing green, rapidly flashing cyan, breathing white, ...?
I do know (most of) these states too, but where are these doced? Users who haven't grown with the Core might need to find it easily - atleast a link to the respective site should be placed along with the respective `WiFi.*`, `Spark.*` and `SYSTEM_MODE` docs.

hpssjellis, 2015-05-07:
For semi-automatic mode please adjust the docs to state that spark.connect() is NOT a blocking call and also insert this code as an example. 

Note: When wifi is connected it breaths Cyan, when Wifi is not connected it breaths Blue.

SYSTEM_MODE(SEMI_AUTOMATIC);
void setup() {
   Spark.connect();   // do not miss this line or you can't flash more code
}
void loop() {
}

Slightly better code would look sort of like:

SYSTEM_MODE(SEMI_AUTOMATIC);

void setup() {
   Spark.connect();  // Very important or you can't reflash your core without a reset

}

void loop() {
    // put your generic code here that works if Wifi connected or not

    if (Spark.connected()){  
            // only put code here that needs Wifi Cloud connectivity
    } 
    
}

closed 2015-09-24 by kennethlimcp